imrahil/OctoPrint-NavbarTemp #1: Could Only Install As Root

eboston, 2015-02-08:
The instructions didn't say it, but the only way I could get the plugin installed was by running it as root (sudo pip ...).  If I didn't use sudo, it reported...

error: could not create '/usr/local/lib/python2.7/dist-packages/octoprint_navbartemp': Permission denied

Is this a problem with the install or does the instructions need to be update or clarified?

imrahil, 2015-02-08:
I copied these instructions from other plugins :) 
using sudo (root) is standard procedure when you want to install something system-wide. So there is no need to mention about that.

closed 2015-02-08 by imrahil

eboston, 2015-02-08:
I guess it wasn't clear to me about it being something system wide as everything else directly related to OctoPrint using pip didn't need to be done using root.