imrahil/OctoPrint-NavbarTemp #3: Could not install on OctoPi

Helisinus, 2015-04-05:
Hi , i don't know if my problem is a Issue because i am new on this. I Use Octopi Version: 1.2.0-dev-705-ga34cbc5 (devel branch).
If i try to install the Plugin with:
pip install https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip

i get this output:
Downloading/unpacking https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
  Downloading master.zip
  Running setup.py egg_info for package from https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip

Downloading/unpacking OctoPrint (from OctoPrint-NavbarTemp==0.3)
  Could not find any downloads that satisfy the requirement OctoPrint (from OctoPrint-NavbarTemp==0.3)
No distributions at all found for OctoPrint (from OctoPrint-NavbarTemp==0.3)

After this, nothing is installed. 
So what have i to do  to install the plugin.

Joerg

imrahil, 2015-05-08:
Try this:
```
source ~/oprint/bin/activate
```
and then: pip install ...

closed 2015-05-08 by imrahil

arekm, 2015-05-13:
Doing
````pip install --upgrade https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip````
has similar problem. Fortunately there is a workaround:
````
pip uninstall OctoPrint-NavbarTemp
pip install https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
````

imrahil, 2015-05-13:
/oprint/bin/pip install --ignore-installed --force-reinstall --no-deps https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip

added info to readme file

arekm, 2015-05-13:
If some files will be gone or renamed in future version then old versions will stay using that command, right? Uninstall & install looks cleaner to me.

imrahil, 2015-05-13:
output from shell:
```
Downloading/unpacking https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
  Downloading master.zip (unknown size): 25Kb downloaded
  Running setup.py egg_info for package from https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip

Installing collected packages: OctoPrint-NavbarTemp
  Found existing installation: OctoPrint-NavbarTemp 0.4
    Uninstalling OctoPrint-NavbarTemp:
      Successfully uninstalled OctoPrint-NavbarTemp
  Running setup.py install for OctoPrint-NavbarTemp

Successfully installed OctoPrint-NavbarTemp
Cleaning up...
```
as you can see there is uninstall & install

arekm, 2015-05-13:
Thanks, new feature works fine here.