imrahil/OctoPrint-NavbarTemp #5: temperature not updated when heating "manually"

arekm, 2015-05-15:
Current temperature doesn't get updated if we tell octoprint to heat hotend (without actually printing). So Temperature -> Set , choose some profile like PLA for hotend. Hotend starts to heat up but temperature shown by navbartemp is not being updated.

imrahil, 2015-05-15:
c'mon! that's the core functionality for this plugin! :) it works! I'm using it all the time!
do you have temperature readings below graph? if yes, there should be the same values on top nav  bar..

arekm, 2015-05-15:
Hehe, works when print is in progress. Looks like OctoPrint problem since graphs and text values in Temperature tab are also not being updated :-/

Closing, will report to the proper place.

closed 2015-05-15 by arekm