imrahil/OctoPrint-NavbarTemp #8: Feature request: select which hotends to display

gertmenke, 2015-07-31:
It would be nice to be able to select which hotends' temperatures should be shown.

I have a Diamond Hotend which means that I have three extruders which share one nozzle and one heater. So I really only need to see one temperature, because they are always the same.
Other people might want to see only the temperature of the currently active extruder even if they don't share a heater.

So the config page might look like this:

    [ ] Raspberry Pi internal temperature
    [x] Bed
    ( ) All hotends
    (o) The currently active hotend
    ( ) Selected hotends
        [x] Hotend 1
        [x] Hotend 2
        [ ] Hotend 3

MennoLauwerens, 2016-04-13:
With the Diamond hotend and Repetier firmware I have 16 virtual extruders (color mixing), I have Octoprint setup for 1 extruder, and made custom commands to select the right one for manual operation.
It would be nice to have native support for this configuration.

LazeMSS, 2021-08-22:
This is possible with my plugin: 
https://plugins.octoprint.org/plugins/navbartemp/

Maybe give it a try?

closed 2023-03-24 by fnordsh