imrahil/OctoPrint-NavbarTemp #12: Request: Option to display build chamber temperature in the navbar

david-vo, 2016-03-24:
I think it would be useful to have the ability to display build chamber temperature in the header, along the same row as Raspi, Tool 0, Tool 1, and Bed temperatures. It can take this value from a thermistor or some other temp sensor, that's fed through a GPIO pin. How difficult would this be to implement?

Seems like it'd be pretty simple to read in the temp from sys/bus?

imrahil, 2016-03-25:
it should be possible the same way as Raspberry internal temperature is measured.

Cosik, 2018-10-16:
@david-vo please look here https://github.com/imrahil/OctoPrint-NavbarTemp/issues/16 also we are thinking about adding nativ support for 1-wire in the future,

Cosik, 2018-10-27:
@david-vo Hi we prepared new feature for custom commands, please read this https://github.com/imrahil/OctoPrint-NavbarTemp/pull/52

Cosik, 2020-03-14:
Custom command is now supported, number of custom commands will be increased.

closed 2020-03-14 by Cosik