imrahil/OctoPrint-NavbarTemp #15: cosmetical: padding of element 'navbar_plugin_navbartemp'

gretel, 2017-01-29:
if `NavbarTemp` is not the leftmost element it will *touch* the adjacent one.
adding `#navbar_plugin_navbartemp { padding-left: 10px }` to `static/css/navbartemp.css` makes me happy 👁 
[image]
[image]

imrahil, 2017-01-31:
PR is welcome :)

Cosik, 2018-10-18:
@akurz42 so it was fixed in past? Should we change css?

akurz42, 2018-10-18:
It definitely looked different (read: better) before the last update. I have two printers here, one updated and one still on the older version. This is what they look like:

v0.9:
[image]

v0.11:
[image]

So something in the formatting definitely seems to have changed between the two versions.

Cheers,
Alex

Cosik, 2018-10-19:
ok, so for good understanding, you want more spacing to get nicer overview?

akurz42, 2018-10-19:
Hi Cosik,

Well, not exactly. I can see that the spacing between the actual values got a bit tighter, too, but that I can live with. The issue is the padding (or lack thereof) at the beginning and the end of the whole temperature line. In my case it's squashed right against the trashcan icon on the right. For the original poster it was the "12 V" symbol at the left. So basically it's this bit (good):
[image]

vs this bit (not so good):
[image]

In general I liked the old, wider spacing better because it's easier to read, but thats no biggie. But there definitely should be some padding at the left & right. I realize that the other plugin ("DeleteAfterPrint" in my case) is making the exact same mistake, but I went here because NavbarTemp had it right before the last update.

Cheers,
Alex

bolsoncerrado, 2022-11-17:
2018, wow! Time flies!

I came here to request more spacing between the elements too as now with the MakeMoreSpaceOnNavBar its also harder to read the temps.

AND perhaps adding BOLD to E:  B: too.