imrahil/OctoPrint-NavbarTemp #16: Enhancement Please - Raspi CPU %

Eddiiie, 2017-02-05:
Hello,
Love your plug in.   Is there any chance you could add one more item to it and that would be the current CPU usage of the Raspi?

I often wonder what the under-lying CPU usage is.

Thank you for your consideration.

imrahil, 2017-02-06:
<lazy-mode>what's the command to check CPU usage? </lazy-mode> :)

Eddiiie, 2017-02-07:
Hi,

I found and tested a script on Octoprint / Raspberry Pi.

Taken from this URL:  http://askubuntu.com/questions/450045/get-cpu-usage-in-command

It seems to work well.

Thank you for your consideration!

[https://cdn.sstatic.net/Sites/askubuntu/img/apple-touch-icon@2.png?v=c492c9229955]<http://askubuntu.com/questions/450045/get-cpu-usage-in-command>

server - get cpu usage in command - Ask Ubuntu<http://askubuntu.com/questions/450045/get-cpu-usage-in-command>
askubuntu.com
To get cpu usage, best way is to read /proc/stat file. See man 5 proc for more help. There is a useful script written by Paul Colby i found here

Text:

To get cpu usage, best way is to read /proc/stat file. See man 5 proc for more help.

There is a useful script written by Paul Colby i found here<http://www.mail-archive.com/linuxkernelnewbies@googlegroups.com/msg01690.html>

#!/bin/bash
# by Paul Colby (http://colby.id.au), no rights reserved ;)

PREV_TOTAL=0
PREV_IDLE=0

while true; do

  CPU=(`cat /proc/stat | grep '^cpu '`) # Get the total CPU statistics.
  unset CPU[0]                          # Discard the "cpu" prefix.
  IDLE=${CPU[4]}                        # Get the idle CPU time.

  # Calculate the total CPU time.
  TOTAL=0

  for VALUE in "${CPU[@]:0:4}"; do
    let "TOTAL=$TOTAL+$VALUE"
  done

  # Calculate the CPU usage since we last checked.
  let "DIFF_IDLE=$IDLE-$PREV_IDLE"
  let "DIFF_TOTAL=$TOTAL-$PREV_TOTAL"
  let "DIFF_USAGE=(1000*($DIFF_TOTAL-$DIFF_IDLE)/$DIFF_TOTAL+5)/10"
  echo -en "\rCPU: $DIFF_USAGE%  \b\b"

  # Remember the total and idle CPU times for the next check.
  PREV_TOTAL="$TOTAL"
  PREV_IDLE="$IDLE"

  # Wait before checking again.
  sleep 1
done

save it to cpu_usage, add execute permission chmod +x cpu_usage and run:

./cpu_usage

to stop the script, hit Ctrl+c

share<http://askubuntu.com/a/450136>improve this answer<http://askubuntu.com/posts/450136/edit>

markaswift, 2017-05-20:
This would be an awesome feature!

mtmello2003, 2017-12-17:
Agree.! could be great feature.

Cosik, 2018-10-16:
@Eddiiie @Swiftnesses @mtmello2003 what do you think if we will give you possibility to execute custom command in shell and print results on Navbar?

mtmello2003, 2018-10-16:
For sure will be the best option: let the user custom by them self.

Eddiiie, 2018-10-20:
@Cosik That would be pretty cool and include the above script as a default?

Cosik, 2018-10-23:
@Eddiiie I'm not sure if in first releases there will be something like "default"

Cosik, 2018-10-27:
@Eddiiie Hi we prepared new feature for custom commands, please read this https://github.com/imrahil/OctoPrint-NavbarTemp/pull/52

Cosik, 2020-03-14:
This could be done by custom command.

closed 2020-03-14 by Cosik