imrahil/OctoPrint-NavbarTemp #17: [request] cpu temperature for armbian device?

ntoff, 2017-02-19:
/etc/armbianmonitor/datasources/soctemp contains a unitless numerical value for the temperature of the SoC chip, is it possible to add another check along side the check for raspbian?

`pi@woody:~$ cat /etc/armbianmonitor/datasources/soctemp`
`38700`

for a temperature of 38.7 degrees C

ntoff, 2017-02-20:
aaaand working here: https://github.com/ntoff/OctoPrint-NavbarTemp/commit/190174cb682ae90d8daeb9d634e35a36e0d31aaf

extreme hackjob and definitely needs cleaning up. I just as much as assumed that an allwinner cpu = armbian which isn't right 

also for whatever strange reason, the regexp search query just won't work with ^ and $, but with them gone the code at least still works on a raspberry pi 2b.

Cosik, 2018-10-16:
@ntoff polease check latest update ;)

closed 2018-10-23 by Cosik