imrahil/OctoPrint-NavbarTemp #18: Missing Settings dialog; forgetting display Raspi temperature

chrissv, 2017-02-25:
Looking at the code, it looks like there should be some settings dialog for this plugin. I am using Octoprint 1.3.1 and no settings are apparent in the Octoprint "Settings" page.

Additionally, the first time I installed the plugin, the Raspi temperature was displayed. But when I restarted the Octoprint application, the display disappeared (only showing the Hotend and Bed temperatures), and I can't figure out how to get it back.

hypersonicdk, 2017-03-20:
I've got the same problem. How can I fix it? thanks

cristiansavaro, 2017-05-14:
same issue here : the config panel under octoprint settings disappeared and the raspi temp disappeared from the navbar as well. i am using octoprint 1.3.3.

timolow, 2017-07-17:
+1 having the same issue on 1.3.4 fresh install

WarpZoneNL, 2017-07-30:
The templatefile "navbartemp_settings_raspi.jinja2" issn't copied right while installing.
Mine was empty! After filling with content from github and a reboot, it works again!

kain000, 2017-08-12:
I copied the text directly from the navbartemp_settings_raspi.jinja2 github page and overwrote the existing navbartemp_settings_raspi.jinja2, but I still have no menu in the Octoprint settings.

Could you specify exactly what you did to restore your settings menu?

dejanzidar, 2017-08-14:
Hi,
in my case problem was, that I use RPi 3 and plugin did not not recognise it. So I edited file:
_**/home/pi/oprint/lib/python2.7/site-packages/octoprint_navbartemp/__init__.py**_
I added recognition Hardware code for RPi 3 (BCM2835)

[__init__.py.zip](https://github.com/imrahil/OctoPrint-NavbarTemp/files/1222587/__init__.py.zip)

then I ompiled __init__.py file by using:
_**python -m compileall /home/pi/oprint/lib/python2.7/site-packages/octoprint_navbartemp/__init__.py**_

reboot and that is it.

Have a nice day

chrissv, 2017-08-27:
Yes! The fix from @dejanzidar took care of the problem. So it looks like the problem is this plugin doesn't recognize a RasPi 3 (which I also have).

Note in the above comment, the underlines in the file name don't show up (got eaten by the github text formatter). The actual file is  `__init__.py` in all cases, and not `init.py`

chrissv, 2017-08-27:
Note this issue has the same root cause as issue #23

closed 2017-12-05 by imrahil