imrahil/OctoPrint-NavbarTemp #23: raspi temp not showing up on rpi 3 hardware

cristiansavaro, 2017-05-14:
can you please analize and appy the attached patch so we can all benefit from this plugin showing up the raspi 3 hardware temp ?
[patch.txt](https://github.com/imrahil/OctoPrint-NavbarTemp/files/999420/patch.txt)

markaswift, 2017-05-20:
Temp works on my RP3 as standard?

cristiansavaro, 2017-05-26:
mine used to work as well but just stopped working until i applied this patch.

rpgrunwald, 2017-06-03:
I added the three lines of code manualyl and it works now with Octoprint 1.3.4 and a Pi 3b.

pmbroth, 2017-07-18:
where did you add the lines of code?

pmbroth, 2017-07-18:
how do you apply the patch?

pmbroth, 2017-07-18:
figured it out thanks

pmbroth, 2017-07-18:
well I added the 3 lines, still no temp

cristiansavaro, 2017-07-19:
you have to patch the code and zip it up. then delete the old plugin and add the new zip through the octoprint interface. that's how i did it and it works just fine.

travmo, 2017-07-28:
Tried to patch the code then use the interface to install the zip, but got an error. This is what I see:

Installing plugin from uploaded archive...
/home/pi/oprint/bin/python -m pip install /tmp/tmpY46Sux.zip
Processing /tmp/tmpY46Sux.zip
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
IOError: [Errno 2] No such file or directory: '/tmp/pip-CkTGq4-build/setup.py'
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-CkTGq4-build/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

WarpZoneNL, 2017-07-30:
1. edit __init__.py
sudo nano ~/oprint/lib/python2.7/site-packages/octoprint_navbartemp/__init__.py

2. place last 3 rows of the code, this will accept the Pi3 proc!
elif match.group(1) == 'BCM2709':
self._logger.debug("Pi 2")
self.isRaspi = True
**elif match.group(1) == 'BCM2835':
self._logger.debug("Pi 3")
self.isRaspi = True**

3. Restart OctoPrint

travmo, 2017-07-31:
Thanks WarpZoneNL. Was trying to make the edits, then manually add it as a plugin. Totally overlooked just installing the plugin, then editing it.

vegasrambler, 2017-08-11:
I have the Pi 3 b and the Pi temp is not showing. The code above is in the init.py file.

Thanks for any help on this issue.

kain000, 2017-08-12:
Same issue, I also tried WarpZoneNL's other solution, populating  navbartemp_settings_raspi.jinja2 with the code form Github but neither that fix or adding the code to init.py have restored the menu for me.  

Rpi 3, fresh image of Octopi

Tinchus2009, 2017-08-18:
I have the same issue with a raspberry pi 3 B. The file ~/oprint/lib/python2.7/site-packages/octoprint_navbartemp/init.py does not exist in that location, instead there is a __init__.py, is that the file I have to edit and add those 3 lines? I did it but still is not working

chrissv, 2017-08-27:
Note that the github comment formatter is eating the underscores. So references to __init__.py are actually for the file `__init__.py`

twstdpear, 2017-09-09:
Doing a little digging, I found out that depending on the version of Octopi/Raspbian/Debian and more specifically certain kernel versions, the Pi3 chip shows up differently, which is why it seems to work for some people and not others.  

- OctoPi 0.14 Jessie, kernel 4.9.35 - shows up as BCM2835
- OctoPi 0.12(?) Wheezy, kernel 4.1.19 (needs updating), shows up as BCM2709
- Raspbian - Jessie, kernel 4.4.21 (hasn't been on in a while), shows up as BCM2709
- Raspbian - Jessie, kernel 4.9.35 (same Pi as previous line, but up to date), shows up as BCM2835

I'm running a job on the printer attached to my Wheezy Pi3, otherwise I'd updated it to see what happens there.  However, it looks like BCM2835 is the "new" id for it in the more recent kernels.

Related discussion here:
https://raspberrypi.stackexchange.com/questions/840/why-is-the-cpu-sometimes-referred-to-as-bcm2708-sometimes-bcm2835

Summary: There seems to be a great amount of confusion regarding which is the chip and which is the package, and without signing an NDA and giving up our ability to disclose that information, we may never know.

closed 2017-12-05 by imrahil