imrahil/OctoPrint-NavbarTemp #26: Themeify + NavbarTemp - text not visable

dragonnn, 2017-11-27:
They is a nice plugin called Themeify with allows you to customize look of Octoprint. Nice black theme called Discorded.  Unfortunately it doesn't work good with NavbarTemp - text has the same color as navbar color is. 
I tried add an option in it to change . navbar-text color but this only changes color of Bed temp, Tool stays at the old color.

mstofflett, 2018-04-21:
Is there a fix for this?

Peboli, 2018-09-09:
Yes there is a workaround.
In the Themeify setup page, click on 'Advanced options' text to unfold advanced settings, under 'Custom style' click  on the [NEW] button and fill it like this
[image]
Then click on [SAVE]

Nuttavoot, 2018-10-09:
> Yes there is a workaround.
> In the Themeify setup page, click on 'Advanced options' text to unfold advanced settings, under 'Custom style' click on the [NEW] button and fill it like this
> [image]
> Then click on [SAVE]

This works only the first time after clicked setting again it disappeared only Bed temp stayed visible.

eddiePower, 2021-01-01:
I found i needed to use the following setting to uncover the text covered up.

[image]

Selector: .socInfo
CSS-Rule: padding
Color Picker: leave as is
Value: 60, 10

that gives the area around the onboard SOC temp for me the chars "Pi: 50.6C" if you need more room add to the first value 60. and reduce that value if you need less space.

Hope it helps till the fix is pushed.

LazeMSS, 2021-01-06:
Fixed in latest release

closed 2022-08-16 by arekm