imrahil/OctoPrint-NavbarTemp #29: Request: short text option

mtmello2003, 2017-12-17:
Option to save navbar space, with short text like "Rpi" (Raspi), "E1" (Tool), delete some spaces and remove "oC"

Thanks!

robertwallace, 2019-02-20:
Agreed - Navigation bar space in Octoprint is limited and fixed and if you cram too much into it, it goes to additional lines.   I generally use long names for my files, including info about the object.  These long names really screw up the NavbarTemp display.  When NavbarTemp starts updating, it often bounces the navbar to two lines, sometimes oscillating back and forth.    Would love the option to leave off the file name - it's already displayed on the screen in the State box, anyway.

Multiple display options would be ideal in any case.  Sometimes I'd like to reduce it to just current bed and nozzle temps without the heating/cooling indicators.

RafaelSchridi, 2020-03-14:
I like the new shorter names but marlin doesn't support multiple beds so b0 could just be bed. and I think E0 is used more than than T0.

But I think the open pullrequest solves everybody's problems.

Cosik, 2020-03-15:
@RafaelSchridi I will change it to "E" and 0 from bed will be removed.

closed 2020-03-15 by Cosik