imrahil/OctoPrint-NavbarTemp #31: [Enhancments] Add Fan Speed

GingerNutt, 2017-12-27:
The title says it all, could you add the fan speed please?

Yours Roy
Computer Techy
PC, MAC, Laptop Repairs
Unlimited Automatic Cloud Backup
www.computertechy.co.uk
BCWD Ltd
Bee Creative Web Design
www.bcwd.ltd.uk
07717454441

Cosik, 2018-10-15:
@GingerNutt could it be as an option in config? @imrahil What do you think about it?

imrahil, 2018-10-15:
How it could possible? Is there any readings with fan speed?

ntoff, 2018-10-23:
https://github.com/ntoff/OctoPrint-FanSpeed

Fan speed is read from gcode sent *to* the printer, and also responses *from* the printer, if it's running repetier firmware. I don't know about marlin or klipper or teacup as I don't run those firmwares.

Cosik, 2018-10-23:
I don't have repetier firmware to test it. Maybe we will think in the future what to do with it.

closed 2024-09-17 by imrahil