imrahil/OctoPrint-NavbarTemp #33: Temperatures not displayed - #settings_plugin_navbartemp does not exist

martinj, 2018-04-05:
The following output can be seen in the console.

> Did not bind view model NavbarTempViewModel to target #settings_plugin_navbartemp since it does not exist

Octoprint version 1.3.6

martinj, 2018-04-05:
Wasn't connected to the printer. Doh!
Once I connected it showed up, though it doesn't seem to register the rapsberry temp, might be due to being the latest raspberry 3 B+

imrahil, 2018-04-05:
I think it's because 3 B+ has new processor ID

martinj, 2018-04-05:
Actually I copied and paste the code for the detection, the hardware still matches.
I was mixing things up. It still exists a problem, its the settings view thats not show and i can't enable show raspberry temp.

Yury-MonZon, 2018-04-24:
Same here =( I see the hotend temp in the navbar, but couldn't find the settings.

Cosik, 2020-03-14:
@martinj could we close this issue or some action are needed?

closed 2020-03-16 by martinj