imrahil/OctoPrint-NavbarTemp #35: NavbarTemp + Themeify = Can't see text

OutsourcedGuru, 2018-06-01:
If you have `Themeify` also loaded and it's using the default color scheme, it appears that the text doesn't display for your plugin.

There is a work-around by adding a custom CSS style in Themeify itself:

```
Selector:  .navbar-text
CSS-Rule:  color
Value:     #ffffff
```

Nuttavoot, 2018-10-09:
> If you have `Themeify` also loaded and it's using the default color scheme, it appears that the text doesn't display for your plugin.
> 
> There is a work-around by adding a custom CSS style in Themeify itself:
> 
> ```
> Selector:  .navbar-text
> CSS-Rule:  color
> Value:     #ffffff
> ```

After click on Octoprint setting again, the Tool temp disappeared again.

Cosik, 2020-03-14:
@Nuttavoot @OutsourcedGuru  this is related to #64. Please pass there information how to fix it if you know how.

closed 2020-03-14 by Cosik