imrahil/OctoPrint-NavbarTemp #43: Wrong name for RPi3 on NavbarTemp

phcoetzee, 2018-10-16:
NavbarTemp used to show the prefix to temp for my RPi3 as Raspi, but today after the update it show the prefix temp as SoC. The temp percentage is correct its just showing: SoC: 35.4°C instead of Raspi: 35.4°C

Cosik, 2018-10-16:
@phcoetzee it's not a bug. We change displayed name to general name.

phcoetzee, 2018-10-16:
Thx, but how can I change Soc to Raspi?

Cosik, 2018-10-16:
@phcoetzee adn @imrahil ofc, we could try to do this, but this should be next feature called something like "custom names in NavbarTemp". There is probably some other FR for that.

phcoetzee, 2018-10-16:
Yes please! :+1:

SillyHats, 2018-12-01:
This would be a good option to add to the settings

Cosik, 2020-03-18:
@phcoetzee and @SillyHats I added such possibility. Please check latest devel and let me know if it's working correctly.

phcoetzee, 2020-03-18:
If I add the name Raspi in the Name Configuration for custom commands it reads:
SoC: 34.3°C Raspi: null in the Navbar so I guess something not right?

Cosik, 2020-03-18:
@phcoetzee are you sure that you have installed plugin from devel branch and you have clean cache of web browser?

phcoetzee, 2020-03-19:
Sorry I have installed the master. I have now installed the devel and its now working perfectly showing custom name. Thank you for your effort.

Cosik, 2020-03-19:
@phcoetzee change will be in official release 0.14 which should be available soon.

closed 2020-03-19 by Cosik