imrahil/OctoPrint-NavbarTemp #44: plugin stopped working after 0.11 update

sigurg, 2018-10-16:
I just updated NavbarTemp to version 0.11 and it seems this release is broken with my octoprint install.

Log shows the following error while loading:

> 2018-10-16 17:12:27,324 - octoprint.server.views - ERROR - Error while retrieving template data for  plugin navbartemp, ignoring it
> Traceback (most recent call last):
>  File "/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/views.py", line 635, in fetch_template_data
>    configs = implementation.get_template_configs()
>  File "/venv/local/lib/python2.7/site-packages/octoprint_navbartemp/__init__.py", line 82, in get_template_configs
 >   if self.sbc.is_supported:
> AttributeError: 'NoneType' object has no attribute 'is_supported'

Octoprint version is 1.3.9 on Debian (testing, x86_64).

Cosik, 2018-10-16:
@sigurg what is your hardware platform? Do you use armbian monitor? How do you read temperature manually from command line?

sigurg, 2018-10-16:
Hardware is standard PC (core i7), thus I have no armbian monitor or anything. I'm only interested in printer bed/nozzle temp, which used to work

Cosik, 2018-10-16:
Ok I will check it asap. Please expect message with test request.

imrahil, 2018-10-16:
@cosik I think that changes for armbian platforms broke something for Windows and/or Linux...

Cosik, 2018-10-16:
@sigurg please uninstall plugin and check it from this source https://github.com/Cosik/OctoPrint-NavbarTemp/archive/bug-fix/normal-pc-support.zip
And then please comment results here:
https://github.com/imrahil/OctoPrint-NavbarTemp/pull/45

Turtletrumpet, 2018-10-16:
Hello,

I had the same problem after upfter to 0.11 like sigburg. I installed manual via https://github.com/Cosik/OctoPrint-NavbarTemp/archive/bug-fix/normal-pc-support.zip and all is fine!!

Cosik, 2018-10-16:
@Turtletrumpet thanks for testing. @imrahil what now? we should make new release? Or how we should provide bug-fixes?

imrahil, 2018-10-16:
From now all new installations will be using fixed code. For people who upgraded to 0.11 reinstallation should help.

Next release should contain better fix for non-arm platforms.

sigurg, 2018-10-16:
Sorry for the delay, my machine was printing so I could not test your fix earlier.
I just re-installed the plugin and can confirm your fix is working on my system.
Thank you for the quick fix!

Cosik, 2018-10-16:
ok, so we could close it.

closed 2018-10-16 by Cosik