imrahil/OctoPrint-NavbarTemp #48: Extruder Temp not updating

joeellis1, 2018-10-20:
Pluging is not updating the extruder temps, alwasys showing 0.  the heated bed shows target and current temps.

Cosik, 2018-10-21:
@joeellis1 please show debug logs and information form terminal when your octoprint is connected to printer.
Another question, was this plugin working on your current printer?

Siggi0902, 2018-10-22:
Habe das selbe Problem seit dem update wird die Tool Temperatur nicht mehr angezeigt

Cosik, 2018-10-23:
@Siggi0902 jest mi niezmiernie miło że znasz inne języki niż angielski ale to nam nie pomoże.

@joeellis1 and @Siggi0902 we didn't get from you any information which could help us debugging a problem. We tested it on different setups and it's working. Please try uninstall and install plugin. If it will don't help we are not able to help you without any debug logs and other additional informations.

DomLuu, 2018-10-26:
Same here ... after a reinstall of the plugin it works a few days ... today tool = 0.0 again. What additional informations you need COSIK and where I can get the debugLogs?

DomLuu, 2018-10-26:
[image]

DomLuu, 2018-10-26:
[image]

The Logs shows 100000. of SoC Temp Checks :)

---->
2018-10-26 08:40:53,592 - octoprint.plugins.navbartemp - INFO - Broadcom detected
2018-10-26 08:40:53,720 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:40:53,793 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2018-10-26 08:40:53,962 - octoprint.plugins.softwareupdate - ERROR - Error while retrieving update information from plugin octoprint_eta
Traceback (most recent call last):
File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/init.py", line 111, in _get_configured_checks
hook_checks = hook()
File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eta/init.py", line 65, in get_update_information
displayName=self._plugin_name,
NameError: global name 'self' is not defined
2018-10-26 08:40:53,983 - octoprint.plugins.stlviewer - INFO - STL Viewer loaded!
2018-10-26 08:40:54,282 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2018-10-26 08:40:54,322 - octoprint.server.util.sockjs - INFO - New connection from client: 85.31.159.242
2018-10-26 08:40:55,300 - octoprint.plugins.excluderegion - INFO - Event received: event=ClientOpened payload={'remoteAddress': 'xxxxxxxxxxxxxxxx'}
2018-10-26 08:40:56,987 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=de', 'path': '/', 'base_url': 'http://octoprint.xxxxxxxxxxxxxxxxxxx.com/'}
2018-10-26 08:41:23,754 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:41:33,484 - octoprint.server.preemptive_cache - INFO - ... done in 36.50s
2018-10-26 08:41:53,806 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:42:23,845 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:42:53,885 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:43:23,923 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:43:53,963 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:44:24,002 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:44:54,044 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:45:24,084 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:45:54,124 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:46:24,164 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:46:54,203 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:47:24,244 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:47:54,284 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:48:24,322 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:48:54,363 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:49:24,403 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:49:54,441 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:50:24,479 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:50:54,520 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:51:24,559 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:51:54,598 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:52:24,636 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:52:54,677 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:53:24,717 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:53:54,757 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:54:24,796 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:54:54,836 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:55:20,400 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-10-26 08:55:24,880 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:55:54,924 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:56:24,962 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:56:55,001 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:57:25,040 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:57:55,079 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 08:58:25,119 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature

Cosik, 2018-10-26:
Here on a picture you have information about log level 
https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299
Please show output from printer Terminal.

Siggi0902, 2018-10-26:
Das steht in meinem Log

2018-10-26 20:11:06,717 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:11:06,763 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2018-10-26 20:11:06,763 - octoprint.server - INFO - OctoPrint 1.3.9
2018-10-26 20:11:06,765 - octoprint.plugin.core - INFO - 20 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Bed Visualizer (0.1.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  DeleteAfterPrint Plugin (1.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DeleteAfterPrint
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Fan Speed Control (0.1.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fanspeedslider
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  LayerDisplay (0.4.1) = /home/pi/oprint/local/lib/python2.7/site-packages/layerdisplay
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  MultiCam (0.2.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_multicam
|  Navbar Temperature Plugin (0.11) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  Octolapse (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Room Temperature Plugin (1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_roomtemp
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2018-10-26 20:11:06,777 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: unknown
|      revision: a020d3
|      version: 0.15.1
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2018-10-26 20:11:06,777 - octoprint.server - INFO - ------------------------------------------------------------------------------
2018-10-26 20:11:36,795 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:12:06,824 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:12:36,851 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:13:06,878 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:13:36,906 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:14:06,988 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:14:37,057 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:15:07,109 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:15:37,137 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:16:03,150 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-10-26 20:16:07,197 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:16:37,268 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:17:07,361 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:17:37,418 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:18:07,476 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:18:37,527 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:19:07,583 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:19:37,635 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 20:19:55,813 - tornado.application - ERROR - Uncaught exception GET /plugin/octolapse/snapshot/latest_snapshot_thumbnail_300px.jpeg?apikey=9CA83B8597D04EB5B60C4C490C86183F&time=1540577997582 (::ffff:192.168.178.20)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1489, in _execute
    result = self.prepare()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 215, in prepare
    self._fallback(self.request, b"")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 604, in __call__
[... cut ...]
2018-10-26 23:40:20,618 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 23:40:50,684 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 23:41:20,735 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 23:41:50,424 - octoprint.util.comm - INFO - Externally triggered heatup detected
2018-10-26 23:41:50,788 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-10-26 23:42:20,816 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature

5ft24dave, 2018-11-11:
I hadn't used my printer in several weeks and noticed the same thing this morning when I fired it up.
I uninstalled the plugin, rebooted the octoprint server, reinstalled it and rebooted the server again and still shows 0.0 as tool temp.  Bed temp and SoC temp are both fine.  No changes to the system or printer since last time it was used.

Cosik, 2018-11-12:
@5ft24dave which language you have set in octoprint? To be honest I really don't know to debug it. I didn't change part responsible for updating temperature of tool and bed.

Cosik, 2018-11-12:
@5ft24dave are you able to make new venv and install in it octoprint and only this plugin and test if problem is visible?

5ft24dave, 2018-11-12:
I will give that a try tomorrow morning.

5ft24dave, 2018-11-14:
I just made a complete new install of Octopi 0.15.1
I get the same result with only the navbar temp plugin.  the only time it will display a tool temperature is if the tool is on and I refresh the browser.  Tested in Chrome, MSIE 11 and MS Edge browsers

Cosik, 2018-11-16:
@5ft24dave sorry for delay. I will find somewhere solution of this problem.

Cosik, 2018-11-25:
last idea, @5ft24dave are you able to show python packages list with versions from yours octoprint env?

5ft24dave, 2018-11-27:
That seemed to do the trick!
Working great now

Cosik, 2018-11-27:
@joeellis1 could you please also udpate you octoprint env libs?

5ft24dave, 2018-12-04:
> @5ft24dave I found some diffs which could be a root cause of problem.
> Please install libs with version from file. Please start from pip,
> setuptools and wheel, then rest.
> [diffpip.txt](https://github.com/imrahil/OctoPrint-NavbarTemp/files/2617025/diffpip.txt)

For the first time in a couple months, I actually needed to print something today.  After updating all these, rebooting octoprint and connecting to the printer, heating bed and extruder multiple times with all of this working last week, the tool temp when I fired it up was back to sitting at 0 again and not showing the commanded temps.  A couple forced reloads with Shift F5, and it will display the temp at that time with the commanded temps as well, but never changes.  I also noticed that the gcode viewer in octoprint is like 6 layers ahead of the printer .. and doing some weird things, like showing it filling in an area, then reversing and "unfilling" it LOL 
I'm wondering if there are some changes made to the November release of Raspbian Stretch that are affecting multiple areas?

⁣Sent from BlueMail ​

Cosik, 2020-03-14:
@joeellis1 and @5ft24dave could we close this issue or problem is still visible?

5ft24dave, 2020-03-14:
> @joeellis1 and @5ft24dave could we close this issue or problem is still visible?

I'm good.  All working as it should

closed 2020-03-18 by Cosik