imrahil/OctoPrint-NavbarTemp #54: X86 Linux CPU Temp

101jallen, 2018-12-14:
Greetings, I have Octoprint on an X86 computer under Ubuntu. Would like to have the CPU temperature on the Navbar. I have found a way to get it there, looking through the source of the NavbarTemp plugin it looks for /etc/armbianmonitor/datasources/sotemp so I created the needed directories and put a symlink named soctemp to  /sys/class/thermal/thermal_zone1/temp it now displays.

I would think it would be fairly straight forward to manage the configuration in your plugin to have a place one could simply put the location of a "file" that it could read with the temperature......

Great plugin as is!

Cosik, 2018-12-15:
Hi,

Sorry last days a lot of happen so I don't have enough time to finish that feature. Meanwhile please look here https://github.com/imrahil/OctoPrint-NavbarTemp/pull/52

Cosik, 2020-03-14:
@101jallen  Custom command are already supported.

closed 2020-03-14 by Cosik