imrahil/OctoPrint-NavbarTemp #57: [Feature Request] Make more room on the navbar

goscicki, 2019-04-21:
The navbar is sometimes very crowded, so it would be nice to have an option to just show the actual temps instead of the actual and target temps. A toggle could allow users to choose.

RafaelSchridi, 2019-10-29:
Yeah a setting would be nice [because I have too much stuff up there](https://i.imgur.com/wRVaXCJ.png) 😅

Cosik, 2020-03-11:
@goscicki and @RafaelSchridi please provide some example or proposal of solution/

RafaelSchridi, 2020-03-13:
This is what I was thinking:
https://i.imgur.com/djKxKKT.png
Hiding the target temperature (not sure what to do with the arrows though, maybe show a different icon if its close?)

Cosik, 2020-03-14:
Hmmm, I will try to make this as optional thing.

Cosik, 2020-03-14:
@RafaelSchridi some of steps has been done, please check #29 you could test it on latest devel branch.

Cosik, 2020-03-18:
@RafaelSchridi I've made additional improvements. Please check if it's ok, or there is still room for improvement.

forthereg, 2020-03-27:
I came here from google looking for this exact thing.
I just need to monitor the Rpi SOC temp as it sits in the heated chamber and this is the ONLY plugin that lets you display Pi temps .
If you could please add option to only display SOC temps, that would be fantastic. 
this is what my bar looks like, makes in into 3 rows  :
[image]

forthereg, 2020-03-27:
I came here from google looking for this exact thing.
I just need to monitor the Rpi SOC temp as it sits in the heated chamber and this is the ONLY plugin that lets you display Pi temps .
If you could please add option to only display SOC temps, that would be fantastic, Thank you.
this is what my bar looks like, makes it into 3 rows, :
[image]

Cosik, 2020-03-27:
@forthereg first please check devel branch if it helps a little bit.

RafaelSchridi, 2020-03-31:
@Cosik Sorry took to long, been a bit hectic lately. But everything looks great on the devel branch, this was exactly what i was looking for. 
Thanks for all the work man!

For the curious this is how it looks now with everything enabled:
[image]

closed 2020-05-07 by Cosik