imrahil/OctoPrint-NavbarTemp #58: cannot identify hardware when running Alpine Linux

draeath, 2019-04-27:
Your method of detecting the hardware is not resilient. I'm running Alpine Linux 3.9 using the Raspberry Pi kernel (4.19.34-0-rpi) and my `/proc/cpuinfo` looks like the following. Note the lack of any "Hardware" lines, which you check for in `__init__.py`.

    processor       : 0
    BogoMIPS        : 38.40
    Features        : fp asimd evtstrm crc32 cpuid
    CPU implementer : 0x41
    CPU architecture: 8
    CPU variant     : 0x0
    CPU part        : 0xd03
    CPU revision    : 4
    
    processor       : 1
    BogoMIPS        : 38.40
    Features        : fp asimd evtstrm crc32 cpuid
    CPU implementer : 0x41
    CPU architecture: 8
    CPU variant     : 0x0
    CPU part        : 0xd03
    CPU revision    : 4
    
    processor       : 2
    BogoMIPS        : 38.40
    Features        : fp asimd evtstrm crc32 cpuid
    CPU implementer : 0x41
    CPU architecture: 8
    CPU variant     : 0x0
    CPU part        : 0xd03
    CPU revision    : 4
    
    processor       : 3
    BogoMIPS        : 38.40
    Features        : fp asimd evtstrm crc32 cpuid
    CPU implementer : 0x41
    CPU architecture: 8
    CPU variant     : 0x0
    CPU part        : 0xd03
    CPU revision    : 4

It is maybe wise to fall back on checking `/proc/device-tree/model` - anyone using recent bootcode will be loading a device tree, and this would then contain useful text. For example, on my RPI 3 B+ it contains `Raspberry Pi 3 Model B Plus Rev 1.3`

Cosik, 2020-05-09:
@draeath do you have something like vcgencmd on Alpine linux?

draeath, 2020-05-09:
I don't recall there's a package for it... But the binaries from the firmware repository work. I keep them in /opt myself.

draeath, 2020-05-09:
See here: https://github.com/raspberrypi/firmware/tree/master/opt/vc