imrahil/OctoPrint-NavbarTemp #59: Cannot Install on ASUS Tinker Board with Linaro (Ubuntu) OS

jalanjarosz, 2019-04-30:
pip install "https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip"

DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.

Collecting https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
  Downloading https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
     / 368kB 4.4MB/s

Collecting OctoPrint (from OctoPrint-NavbarTemp==0.11)
  ERROR: Could not find a version that satisfies the requirement OctoPrint (from OctoPrint-NavbarTemp==0.11) (from versions: none)

ERROR: No matching distribution found for OctoPrint (from OctoPrint-NavbarTemp==0.11)

jalanjarosz, 2019-04-30:
Here is some additional information from Octoprint plug-in log:

```
> > 2019-04-30 21:29:54,532   /home/linaro/OctoPrint/venv/bin/python -m pip install https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip --no-cache-dir
> 2019-04-30 21:29:56,069 ! DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
> 2019-04-30 21:29:56,850 > Collecting https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
> 2019-04-30 21:29:56,851 > Downloading https://github.com/imrahil/OctoPrint-NavbarTemp/archive/master.zip
> 2019-04-30 21:29:58,337 > Requirement already satisfied: OctoPrint in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint-NavbarTemp==0.11) (1.3.10)
> 2019-04-30 21:29:58,337 > Requirement already satisfied: pyserial<3.5,>=3.4 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (3.4)
> 2019-04-30 21:29:58,337 > Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.0.9)
> 2019-04-30 21:29:58,338 > Requirement already satisfied: semantic-version<2.5,>=2.4.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (2.4.2)
> 2019-04-30 21:29:58,338 > Requirement already satisfied: future<0.16,>=0.15.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.15.2)
> 2019-04-30 21:29:58,338 > Requirement already satisfied: scandir<1.4,>=1.3 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.3)
> 2019-04-30 21:29:58,338 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.6.5)
> 2019-04-30 21:29:58,338 > Requirement already satisfied: futures<3.2,>=3.1.1 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (3.1.1)
> 2019-04-30 21:29:58,339 > Requirement already satisfied: rsa<3.3,>=3.2.3 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (3.2.3)
> 2019-04-30 21:29:58,339 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.10)
> 2019-04-30 21:29:58,339 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.7.19)
> 2019-04-30 21:29:58,339 > Requirement already satisfied: tornado==4.5.3 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (4.5.3)
> 2019-04-30 21:29:58,340 > Requirement already satisfied: flask<0.11,>=0.10.1 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.10.1)
> 2019-04-30 21:29:58,340 > Requirement already satisfied: Click<6.3,>=6.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (6.2)
> 2019-04-30 21:29:58,340 > Requirement already satisfied: regex!=2018.11.6 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (2019.4.14)
> 2019-04-30 21:29:58,340 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.0.3)
> 2019-04-30 21:29:58,340 > Requirement already satisfied: wrapt<1.11,>=1.10.11 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.10.11)
> 2019-04-30 21:29:58,341 > Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (3.10)
> 2019-04-30 21:29:58,341 > Requirement already satisfied: monotonic<1.4,>=1.3 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.3)
> 2019-04-30 21:29:58,341 > Requirement already satisfied: frozendict<1.3,>=1.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.2)
> 2019-04-30 21:29:58,341 > Requirement already satisfied: netifaces<0.11,>=0.10.6 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.10.9)
> 2019-04-30 21:29:58,341 > Requirement already satisfied: pkginfo<1.5,>=1.4.2 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (1.4.2)
> 2019-04-30 21:29:58,342 > Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (0.8.3)
> 2019-04-30 21:29:58,342 > Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (2.6.1)
> 2019-04-30 21:29:58,342 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/linaro/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->OctoPrint-NavbarTemp==0.11) (5.2.1)
[... cut ...]
> 2019-04-30 21:29:59,372 ! ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '/home/linaro/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint_NavbarTemp-0.12.dist-info/RECORD'
> 2019-04-30 21:29:59,372 ! 
> 2019-04-30 21:29:59,882 > Building wheel for OctoPrint-NavbarTemp (setup.py): finished with status 'done'
> 2019-04-30 21:29:59,885 > Stored in directory: /tmp/pip-ephem-wheel-cache-vASTMr/wheels/df/3d/05/7c32e9098fc0d557edcff3d9b0667e0abdd69bc5e9682587ef
> 2019-04-30 21:29:59,887 > Successfully built OctoPrint-NavbarTemp
> 2019-04-30 21:29:59,888 > Installing collected packages: OctoPrint-NavbarTemp
> 
> > 
```

Cosik, 2020-03-14:
@jalanjarosz is problem still visible?

jalanjarosz, 2020-03-18:
I ended up switching to Debian Stretch with Armbian Linux 4.19.69-rockchip and have no issues using your plug-in.  Not sure if there is something missing in Linaro, but Armbin is working just fine.

arekm, 2022-08-16:
Please reopen with new data if this no longer works with current version.

closed 2022-08-16 by arekm