imrahil/OctoPrint-NavbarTemp #64: Themeify compatible

pawaller, 2019-10-28:
Is it possible to make your plugin compatible with the "themeify" plugin.
I.e it takes the font color etc from the applied theme.

Thanks

buchnoun, 2019-11-12:
I would love that too !! it was working fine and stopped... very frustrating

ChaosBlades, 2019-11-25:
Ah, so that is why it doesn't work.

Cosik, 2020-03-14:
@pawaller @buchnoun @ChaosBlades are you able to pass information when last it was working with themeify?

iMattmax, 2020-06-01:
[image]
I have the same result with using this (OctoPrint-NavbarTemp) and Themeify

ChaosBlades, 2020-06-01:
I gave up on Themeify. I just use Dark Reader now  which is a browser plugin. Works perfectly and looks way better than the Themeify themes.

https://github.com/darkreader/darkreader

[image]

Cosik, 2020-06-01:
Sorry I don't have enough skills to fix this issue.

iMattmax, 2020-06-01:
> I gave up on Themeify. I just use Dark Reader now which is a browser plugin. Works perfectly and looks way better than the Themeify themes.
> 
> https://github.com/darkreader/darkreader
> 
> [image]

how did you change the order of the tasks (temperature, terminal, etc.)?

ChaosBlades, 2020-06-01:
@iMattmax I think you are looking for the plug-in called 'Dragon Order'

https://github.com/jneilliii/OctoPrint-DragonOrder

djalexz, 2020-07-31:
Hi, you can set the following Settings in Themeify Options:

[image]

The Text in Navbar is now in the color of .navbar-text.

LazeMSS, 2021-01-06:
Fixed in latest release

closed 2022-08-16 by arekm