imrahil/OctoPrint-NavbarTemp #65: [feature request] Show SoC temperature when not connected to printer

rafalmag, 2019-11-05:
Would it be possible to show the SoC temperature when the printer is not connected?
Current workaround for me is to use [Opitemp Plugin](https://github.com/hashashin/OctoPrint-OpiTemp), but as your plugin supports various boards (including Raspberry Pi) I believe it could be useful for wider audience. 
Motivation for this usecase:
I keep OrangePi in the same enclosure as the printer board, however I switch on/off printer mainboard (and thus its fan) separately. When the fan is on (printer baord is powered) the SoC is around 25-30*C (even during printing!), but when it is off - the SoC goes up to 40-60*C.
I would like to be more aware of this.

Thorinair, 2020-02-17:
+1 for this, I have a system which turns on the Raspberry Pi fans only when the whole printer is on, and I would like to live compare the temperatures even when the printer is off.

Cosik, 2020-03-11:
@rafalmag and @Thorinair do you know how to do it? I was diging in plugin but I was not able to change it.

Cosik, 2020-03-13:
@rafalmag I found how to do it, but all temp on the same time or nothing

closed 2020-03-14 by Cosik