imrahil/OctoPrint-NavbarTemp #71: Raspi Temp not visible

arut16, 2020-05-06:
Hi everyone !

Just installed my fresh new RPi 4 with last Octoprint and NavBarTemp Plugin. Working well for bed/tool temp but it not showing RPi Soc Temp. Why ?

[image]

Cosik, 2020-05-07:
Have  no idea. You didn't provide any information. Why?

arut16, 2020-05-07:
@Cosik witch information do you need ? Sorry I'm a beginner.

Cosik, 2020-05-07:
Please enable full logging for plugin and attache to this issue. Also few minutes ago I release 0.14 version. Please check it first.

arut16, 2020-05-07:
@Cosik I updated to 0.14 but SOC temp still not showed : 
[image]
And for logs, which one do I need upload here ? 
[image]

Cosik, 2020-05-07:
1. https://docs.octoprint.org/en/master/bundledplugins/logging.html
2. If something is not working you could use custom commands.
3. Please carefully read https://github.com/imrahil/OctoPrint-NavbarTemp/blob/master/README.md
4. I have no rpi4 to test it.

arut16, 2020-05-07:
@Cosik Thanks for helping :
1 : I read it. Here logs : https://gist.github.com/arut16/d2f934afa5e332d664a8552556a7faf9
2  : What do you meen custom commands ? How ? Where ?
3 : Read it. Didn't help...
4 : Should be same as Pi3 at this point no?

Cosik, 2020-05-08:
You answers didn't help. Log is still not correct. In readme you have information about custom commands and what to do when platform is not supported. I uploaded issue template. Please fill it.

**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**

**2. Octoprint version:**

**3. Plugin version:**

**4. Print screen of plugin configuration:**

**5. Clear problem description:**

**6. Expectation:**

arut16, 2020-05-08:
**1. Hardware and operating system version:**

Raspberry 4 2Go RAM version
Octopi v.0.17.0

**2. Octoprint version:** v1.4.0

**3. Plugin version:** 0.14

**4. Print screen of plugin configuration:** 
[image]

**5. Plugin full log file** : Don't know how to do that...

**6. Clear problem description:** RPi SOC Temp is not appears

**7. Expectation:** RPi SOC Temp Appears with tool and bed temp.

arut16, 2020-05-08:
I find a solution with a custom command : 
[image]
[image]

closed 2020-05-08 by arut16

Cosik, 2020-05-08:
@arut16 you could add to cmd
` | cut -d'=' -f2'`

arut16, 2020-05-08:
@Cosik Like this : `/opt/vc/bin/vcgencmd measure_temp | cut -d'=' -f2'`

arut16, 2020-05-09:
@Cosik This command work : `/opt/vc/bin/vcgencmd measure_temp | sed -e 's/temp=//'`
With `| cut -d'=' -f2'`, not working
[image]
[image]

dan-and, 2020-09-09:
> cut -d'=' -f2'

It works, but there is just a missing single quote:
cut -d'=' -'f2'

so using:
/opt/vc/bin/vcgencmd measure_temp|cut -d'=' -'f2'
as a custom command helps.