imrahil/OctoPrint-NavbarTemp #73: Heat Bed temp is NaN when NO HEAT BED is used

geodave810, 2020-05-09:
**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**
Raspberry Pi 3B+

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
**2. Octoprint version:**

  Octoprint 1.4.0 on Octopi 0.17.0

**3. Plugin version:**
0.14

**4. Print screen of plugin configuration:**
[image]

[image]

**5. Plugin full log file**
I don't see a log file for this plugin in /home/pi/.octoprint/logs folder

**6. Clear problem description::**
I do not have a Heat bed attached to this printer, but this plugin does not take that into account.  I believe it worked correctly with previous version.  Here is the screen shot of my default printer profile for the heatbed tab.
[image]

**7. Expectation:**
I expect it to leave this field blank when no heat bed is used.

Cosik, 2020-05-09:
@geodave810 you have to enable full logging in octoprint logging configuration. It will help debugging.

LazeMSS, 2021-01-03:
This is fixed and could be closed

closed 2021-01-03 by Cosik