imrahil/OctoPrint-NavbarTemp #74: No SoC temperature

tsndr, 2020-06-06:
**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**
<details>
<summary><code>/proc/cpuinfo</code></summary>

```
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03111
Serial          : 1000000053479ad8
Model           : Raspberry Pi 4 Model B Rev 1.1
```
</details>

<details>
<summary><code>/etc/debian_version</code></summary>

```
10.4
```
</details>

<details>
<summary><code>rpi-eeprom-update</code></summary>

```
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
```
</details>

**2. Octoprint version:**
`1.4.0`

**3. Plugin version:**
`0.14`

**4. Print screen of plugin configuration:**
<details>
<summary>Screenshot</summary>

[image]
</details>

**5. Plugin full log file**
There is no logfile.

**6. Clear problem description::**
If no printer is connected the Navbar shows `Tool: 0.0°C`, but it should show the SoC's temperature.
If a printer is connected it shows `Tool: 21.0°C Bed: 20.6°C` which is fine, but it's still missing the SoC's temperature.

**7. Expectation:**
If no printer is connected the output should be: `SoC: 52.0°C`.
If a printer is connected the output should be: `SoC: 52.0°C Tool: 21.0°C Bed: 20.6°C`

Cosik, 2020-06-06:
@tsndr you need to enable logging. Also it looks like #71

tsndr, 2020-06-10:
How do I do that?

V0lk1s, 2020-07-01:
Could this be related to the problem I found with 64-bit version of Raspberry OS? The command for checking SoC temperature on a Raspberry have a different path.

ManuelMcLure, 2020-07-15:
I've created PR #78 to fix this.

LazeMSS, 2021-01-03:
This is also fixed in current released version if im not mistaken @Cosik ?

Cosik, 2021-01-03:
yes, correct

closed 2021-01-03 by Cosik