imrahil/OctoPrint-NavbarTemp #80: Tool temp shows as 0°C when printer is turned off

timur-tabi, 2020-07-27:
This is a very minor bug.

When the printer is turned on, NavbarTemp shows the SoC, Tool (extruder), and Bed temp just fine:

[image]

However, when the printer is turned off, the Bed temp is not shown (which is good), but the Tool temp is shown as 0°C (which is bad): 

[image]

It's inconsistent that NavbarTemp knows not to show the Bed temperature if the printer is off, but it also should not show the Tool temp if the printer is off.

**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**

RPi 3B+
Linux octopi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

**2. Octoprint version:**

OctoPrint version : 1.4.0
OctoPi version    : 0.15.1

**3. Plugin version:**

0.14

**4. Print screen of plugin configuration:**

[image]

**5. Plugin full log file**

[octoprint.log](https://github.com/imrahil/OctoPrint-NavbarTemp/files/4984723/octoprint.log)

LazeMSS, 2021-01-06:
Should be fixed in latest release

Cosik, 2021-01-10:
@timur-tabi did you check it?

timur-tabi, 2021-01-10:
I'm still on 0.14.  How do I test a pre-release version of your plugin?

closed 2024-09-17 by imrahil