imrahil/OctoPrint-NavbarTemp #81: Navbar Temp not showing on raspberry pi 3b+

Jonijakob, 2020-07-29:
hi I am trying to check the temperature on the raspberry pi 3b+ and it didn't show it
Maybe because the Broadcom is BCM2837B0?

bepstein111, 2020-08-01:
Same here. Using `/opt/vc/bin/vcgencmd measure_temp` as the command works fine but adds an ugly temp= before it. I'm not great with awk, but I think the following command is correct as it works in terminal, but not in Navbar: `/opt/vc/bin/vcgencmd measure_temp | awk -F= '{printf("%d%s\n", $2,  "°C")}'`

rknobbe, 2020-08-05:
Same here with Raspberry Pi 4

robertomilan, 2020-08-07:
Rasperry Pi 4 1GB on OctoPrint 1.4.1/1.4.2, SoC not shown here too.

Anubisbe, 2020-08-08:
same problem no more Soc temp after 1.4.2 release of octoprint and update of raspberry (Raps pi 4B) python2.7
try the fix ""if sys.platform.startswith("linux"):"" but no change

DodgeDeBoulet, 2020-08-08:
I did a little digging in the source and believe I have located the issue. I've fixed it in my local installation and it appears to work for the Raspberry Pi 4.

Line 17 of ~/oprint/lib/python2.7/site-packages/octoprint_navbartemp/libs/sbc.py is:

```
    piSocTypes = (["BCM2708", "BCM2709", "BCM2835"])
```
And should be changed to:

```
    piSocTypes = (["BCM2708", "BCM2709", "BCM2835", "BCM2711"])
```
Line 104 is:
```
         self.temp_cmd = '/opt/vc/bin/vcgencmd measure_temp'
```
And **may** need to be changed to:

```
         self.temp_cmd = '/usr/bin/vcgencmd measure_temp'
```
The reason I say "may" is that vcgencmd exists in both locations and appears to produce the same output. I read another post on a related issue that indicated that the path may need to be updated, because changing it for that user fixed his/her issue.

ChaosBlades, 2020-08-13:
Works for me on Pi4 and OctoPi 0.17.0

I only needed to add the new piSocType.

robertomilan, 2020-08-13:
Works on Pi4 1GB OctoPi 0.17.0. Changed piSocType only.

domhardt, 2020-10-10:
Same problem here. Raspberry Pi 3B+, OctoPrint 1.4.2 running on OctoPi image, Navbar Temperature Plugin 0.14, Python 3

rippiedoos, 2020-10-28:
My best guess is line 38, as mentioned in #78
```
if sys.platform == 'linux2':
```
If you're running Python3, the SBC-code never get's called because of the undetected platform. I changed that line to:
```
if sys.platform.startswith('linux'):
```
And the SoC-temp get's displayed again.

LazeMSS, 2021-01-03:
This is fixed in the current released version and should be closed :)

closed 2021-01-03 by Cosik