imrahil/OctoPrint-NavbarTemp #82: Please add BCM2711 to the SoC list for Raspberry Pi 4

Alliex, 2020-10-26:
**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**
Raspberry Pi 4 B 2Gb
Raspbian 10 (Buster)

**2. Octoprint version:**
Version 1.4.2

**3. Plugin version:**
.14 / latest

**4. Print screen of plugin configuration:**
[image]

**5. Plugin full log file**
N/A

**6. Clear problem description::**
The SoC temp doesn't show up by default.

**7. Expectation:**
Please add 'BCM2711' to piSocTypes ( https://github.com/imrahil/OctoPrint-NavbarTemp/blob/0efd791a590b27b4ca281422be8395e6d1ff85a7/octoprint_navbartemp/libs/sbc.py#L17 ) for the Raspberry Pi 4. Thanks!

rippiedoos, 2020-10-28:
In issues #81 the CPU-code isn't called. Just change around line 40 https://github.com/imrahil/OctoPrint-NavbarTemp/blob/0efd791a590b27b4ca281422be8395e6d1ff85a7/octoprint_navbartemp/__init__.py#L40
To:
```
if sys.platform.startswith("linux"):
```

And you're good to go, same as PR #78 .

LazeMSS, 2021-01-03:
This can be closed - its all fixed in the current version as far as I can tell:
https://github.com/imrahil/OctoPrint-NavbarTemp/blob/master/octoprint_navbartemp/__init__.py#L40
https://github.com/imrahil/OctoPrint-NavbarTemp/blob/master/octoprint_navbartemp/__init__.py#L23

closed 2021-01-03 by Cosik

stepman0, 2022-01-09:
Is there already a new version released? I am still on version 0.14 released in May 2020. Since the upgrade to RPi4 your cool plugin does not show the SoC temperature anymore...