imrahil/OctoPrint-NavbarTemp #85: Make more room in navbar can't be turned off.

Build-0-Matic, 2021-01-17:
OctoPI Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3
OctoPrint Version 1.5.2
Plugin v 0.13

Plugin Configuration
`
General
Configuration for compatible hardware
Display internal temperature
Use short tools names
Make more room in navbar
Configuration for custom commands
SoC name to display
Name
Cmd
`
When the "Make more room in navbar" box is unchecked and that the configuration is saved, the box automatically gets reset to the ON position.

vitorvs88, 2023-05-13:
Lol, it took me a whilr but I found how to solve it. It seems to be a conflict with the plugin UI Customizer. You can make it work by deselecting "Compact Navbar Temp plugin" from the UI Customizer config.

Build-0-Matic, 2023-05-13:
Totally wasn't expecting that one.
I'll have to check it out... had this problem so long ago I kinda forgot about it 😁
Thanks!

closed 2023-05-13 by Build-0-Matic

vitorvs88, 2023-05-15:
the only reason I saw your post is because I had the same problem and was looking for a solution. It was bothering me a lot so I tried a different things to fix it and got lucky. Let me know if it works for you.

Build-0-Matic, 2023-06-14:
> the only reason I saw your post is because I had the same problem and was looking for a solution. It was bothering me a lot so I tried a different things to fix it and got lucky. Let me know if it works for you.

Sorry for taking so long to reply!
I was super busy and didn't have time to start the printer to see if it works.

But yes, it looks like it does. I guess Customize UI can take control of this plugin as when you turn the option off in Navbartemp and re-enable it in Customize UI, the checkmark comes back in Navbartemp. So it would be normal behavior in this case.

Good to know as we can now put this strange issue to rest 😁

Thanks!