imrahil/OctoPrint-NavbarTemp #86: Custom commands clarification / HTS221sensos

alx-nv, 2021-02-25:
**PLEASE CAREFULLY FILL EACH POINT**

**1. Hardware and operating system version:**
Raspi 3b+
adafruit HTS221

**2. Octoprint version:**
Octoprint 1.5.3, octopi 0.18.0

**3. Plugin version:**
v 0.13 ? as of feb 2021

**4. Print screen of plugin configuration:**

**5. Plugin full log file**

**6. Clear problem description::**
I need some info regarding the custom commands - what are those (gcode, ssh..) Is there a list of possible commands? How do I know what commands are legal?

**7. Expectation:**
I am trying to attach a sensor (HTS221, temp and humidity). I have the python script working, I2C etc. Just want to display the values next to the current temps in the plugin.

Thanks!
alx

Cosik, 2021-02-26:
@alx-nv those are linux command line commands as in examples.

alx-nv, 2021-02-26:
Works, thanks!

closed 2021-03-16 by alx-nv